linux-automation/usbsdmux #10: USB-SD-Mux hardware availability

sephalon, 2018-12-11:
Is it possible to buy the USB-SD-Mux hardware as a commercial product? Or can one receive hardware samples for testing purposes (we'd be happy to share our results and contribute bug reports)? If none of this works, would you consider sharing the schematics/gerber files/BOM so we can build one by ourselves?

rschwebel, 2018-12-11:
Hi,

On Tue, Dec 11, 2018 at 07:44:04AM -0800, sephalon wrote:
> Is it possible to buy the USB-SD-Mux hardware as a commercial product? Or can
> one receive hardware samples for testing purposes (we'd be happy to share our
> results and contribute bug reports)? If [...]

We are working on a plan how to sell them, they will probably be
available early next year. I'll put you on my list of interested
persons!

Regards,
Robert
-- 
Pengutronix e.K.                           | Dipl.-Ing. Robert Schwebel  |
Industrial Linux Solutions                 | https://www.pengutronix.de/ |
Peiner Str. 6-8, 31137 Hildesheim, Germany | Phone: +49-5121-206917-0    |
Amtsgericht Hildesheim, HRA 2686           | Fax:   +49-5121-206917-5555 |

SmithChart, 2018-12-12:
I will leave this issue open for reference and close it once our hardware is available.

Chris

sephalon, 2019-02-20:
As we desperately look for a solution to replace our flaky and slow WiFi SD cards (and "early 2019" is/will be over soon, depending on your definition ;-), I'd like to inquire about the current plans for hardware availability?

rschwebel, 2019-02-20:
On Wed, Feb 20, 2019 at 12:56:20AM -0800, sephalon wrote:
> As we desperately look for a solution to replace our flaky and slow WiFi SD
> cards (and "early 2019" is/will be over soon, depending on your definition ;-),
> I'd like to inquire about the current plans for hardware availability?

We are not finished yet, but things are making progress :-)

rsc

ctyler, 2019-03-08:
Count us in as interested too (will also send email).

-Chris Tyler, Seneca CDOT

rschwebel, 2019-03-08:
On Fri, Mar 08, 2019 at 10:09:10AM -0800, Chris Tyler wrote:
> Count us in as interested too (will also send email).

Will do :-)

rsc
-- 
Pengutronix e.K.                           | Dipl.-Ing. Robert Schwebel  |
Industrial Linux Solutions                 | https://www.pengutronix.de/ |
Peiner Str. 6-8, 31137 Hildesheim, Germany | Phone: +49-5121-206917-0    |
Amtsgericht Hildesheim, HRA 2686           | Fax:   +49-5121-206917-5555 |

abelvesa, 2019-06-16:
Any news about this ?

rschwebel, 2019-06-16:
On Sun, Jun 16, 2019 at 11:16:16AM -0700, Abel Vesa wrote:
> Any news about this ?

We are waiting for the production run, stay tuned...

rsc

RobertBerger, 2019-06-17:
@rschwebel  Please put me on the hardware interest list as well ;)

benjaminSchilling33, 2019-08-12:
@rschwebel Please add me to the list of interested people as well :)

rschwebel, 2019-08-12:
On Mon, Aug 12, 2019 at 08:47:55AM -0700, benjaminSchilling33 wrote:
> @rschwebel Please add me to the list of interested people as well :)

Done.

sephalon, 2019-11-12:
Since the hardware is out now, I assume this can be closed?

SmithChart, 2019-11-12:
Totally right. Thanks for the reminder.

closed 2019-11-12 by SmithChart