linux-automation/usbsdmux #11: cannot write configuration EEPROM after first 256 bytes

phipvmatrix, 2019-07-12:
I am using your usb2642eeprom.py command line tool under ubuntu 19.04. Writing of the first 256 bytes works well but my I2C debugger notices that you do not write behind this.

SmithChart, 2019-07-15:
Hi phipvmatrix,
is your EEPROM connected to the auxiliary- or configuration I2C-Bus of the USB2642?

The interaction with the configuration I2C-bus is mostly reverse engineered from the Microchip Config tool for this specific IC. Maybe we have missed something there?

phipvmatrix, 2019-07-15:
My EEPROM is connected to the auxiliary and the configuration I2C-Bus of the USB2642. I can see that the working tool for the other OS is writing to the EEPROM for the first 256 Bytes to the EEPROM Address 0x50 and the rest to Address 0x51. I cannot see that your code can switch the used Address.
Just to be positive: 
The first 256 Byte can be written what is very important and helpful for me. Thank you very much for sharing this code.

SmithChart, 2019-07-16:
Thank your for your feedback. It's good to hear that someone is actually using the low-level interface to the config EEPROM.

For our application we are not using memory locations 0x100 to 0x17F. Thus I guess we never tested that. I currently do not have our USB-analyzer (nor a Windows machine) at hand. 

Can you capture a trace of the USB transfers done by the Microchip tool writing the config EEPROM? Otherwise I will try to get our set-up running sometime in the next days.

Such trace should allow us to see the difference between our implementation and what microchip does.

phipvmatrix, 2019-07-16:
Please let me try to capture this first. I have an analyzer and an installation with Microchip tool. I  need you for the things I do not understand. It is going to take some days. Please expect my reply next Tuesday.

phipvmatrix, 2019-08-04:
Solution

See File: usb2642i2c.py
to write 384 Byte:                                                            x
scsiCommand = list_to_uint8_array([0xCF, 0x54, 0x04, 0x80, 0x00, 0x00, 0x00,\
to write 256 Byte:
scsiCommand = list_to_uint8_array([0xCF, 0x54, 0x04, 0x00, 0x00, 0x00, 0x00,\

phipvmatrix, 2019-08-04:
works for me.
Thank you very much for sharing this code.

closed 2019-08-04 by phipvmatrix