linux-automation/usbsdmux #15: udev rule does not match

sephalon, 2019-11-12:
The provided udev rule does not match our device:

```
$ udevadm info -n /dev/sg2 -a
  looking at device '/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.3/3-1.3.1/3-1.3.1:1.0/host6/target6:0:0/6:0:0:0/scsi_generic/sg2':
    KERNEL=="sg2"
    SUBSYSTEM=="scsi_generic"
    DRIVER==""

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.3/3-1.3.1/3-1.3.1:1.0/host6/target6:0:0/6:0:0:0':
    KERNELS=="6:0:0:0"
    SUBSYSTEMS=="scsi"
    DRIVERS=="sd"
    ATTRS{blacklist}==""
    ATTRS{device_blocked}=="0"
    ATTRS{device_busy}=="0"
    ATTRS{dh_state}=="detached"
    ATTRS{eh_timeout}=="10"
    ATTRS{evt_capacity_change_reported}=="0"
    ATTRS{evt_inquiry_change_reported}=="0"
    ATTRS{evt_lun_change_reported}=="0"
    ATTRS{evt_media_change}=="0"
    ATTRS{evt_mode_parameter_change_reported}=="0"
    ATTRS{evt_soft_threshold_reached}=="0"
    ATTRS{inquiry}==""
    ATTRS{iocounterbits}=="32"
    ATTRS{iodone_cnt}=="0x2e9a"
    ATTRS{ioerr_cnt}=="0x1a1"
    ATTRS{iorequest_cnt}=="0x2e9a"
    ATTRS{max_sectors}=="240"
    ATTRS{model}=="sdmux HS-SD/MMC "
    ATTRS{queue_depth}=="1"
    ATTRS{queue_type}=="none"
    ATTRS{rev}=="2.09"
    ATTRS{scsi_level}=="0"
    ATTRS{state}=="running"
    ATTRS{timeout}=="30"
    ATTRS{type}=="0"
    ATTRS{vendor}=="LinuxAut"

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.3/3-1.3.1/3-1.3.1:1.0/host6/target6:0:0':
    KERNELS=="target6:0:0"
    SUBSYSTEMS=="scsi"
    DRIVERS==""

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.3/3-1.3.1/3-1.3.1:1.0/host6':
    KERNELS=="host6"
    SUBSYSTEMS=="scsi"
    DRIVERS==""

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.3/3-1.3.1/3-1.3.1:1.0':
    KERNELS=="3-1.3.1:1.0"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb-storage"
    ATTRS{authorized}=="1"
    ATTRS{bAlternateSetting}==" 0"
    ATTRS{bInterfaceClass}=="08"
    ATTRS{bInterfaceNumber}=="00"
    ATTRS{bInterfaceProtocol}=="50"
    ATTRS{bInterfaceSubClass}=="06"
    ATTRS{bNumEndpoints}=="02"
    ATTRS{supports_autosuspend}=="1"

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.3/3-1.3.1':
    KERNELS=="3-1.3.1"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{authorized}=="1"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{bDeviceClass}=="00"
    ATTRS{bDeviceProtocol}=="00"
    ATTRS{bDeviceSubClass}=="00"
    ATTRS{bMaxPacketSize0}=="64"
    ATTRS{bMaxPower}=="96mA"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{bNumInterfaces}==" 1"
    ATTRS{bcdDevice}=="0209"
    ATTRS{bmAttributes}=="80"
    ATTRS{busnum}=="3"
    ATTRS{configuration}==""
    ATTRS{devnum}=="8"
    ATTRS{devpath}=="1.3.1"
    ATTRS{idProduct}=="4041"
    ATTRS{idVendor}=="0424"
    ATTRS{ltm_capable}=="no"
    ATTRS{manufacturer}=="Linux Automation GmbH"
    ATTRS{maxchild}=="0"
    ATTRS{product}=="usb-sd-mux_rev4.0"
    ATTRS{quirks}=="0x0"
    ATTRS{removable}=="fixed"
    ATTRS{serial}=="000000000343"
    ATTRS{speed}=="12"
    ATTRS{urbnum}=="25494"
    ATTRS{version}==" 1.10"

  …
```
The label on the device reads:

```
USB-SD-Mux
rev4  Nr.0343
```

And the QR code on the device label reads:

```
0044 30.2019 023800 USBSDMux rev4
```

I assume this udev rule has not been updated yet for devices shipped to customers?

SmithChart, 2019-11-13:
> I assume this udev rule has not been updated yet for devices shipped to customers?

You are right. The manufacturer name changed from the first to the second generation of USB-SD-Mux. I have updated the rules in d6a0c412ab6c69bb3b03fe17e786f2deacd74dc5. Can you give this new udev-rule a try?

sephalon, 2019-11-18:
The new rule matches! Closed.

closed 2019-11-18 by sephalon