linux-automation/usbsdmux #16: Product identification description does not match label

sephalon, 2019-11-12:
The [product identification section in the documentation](https://testbed.linux-automation.com/usbsdmux-M01/product-identification.html#product-identification) does not match the label on our device; it reads:

```
USB-SD-Mux
rev4  Nr.0343
```

And the QR code on the device label does not match either; it reads:

```
0044 30.2019 023800 USBSDMux rev4
```

SmithChart, 2019-11-12:
Thank you for the hint.
This mismatch is true. Our Manufacturer placed labels on the devices that do not match with our internal standard. To identify the correct type please refer to the label attached to the packing.

Please refer to the Serial Number ("Nr. xxxx") printed on the label. This number matches (apart from zero padding) the serial number reported via USB.

I will bring a new version of the documentation the way.

SmithChart, 2019-11-18:
A new version of the documentation is available: https://www.linux-automation.com/usbsdmux-M01/

> And the QR code on the device label does not match either; it reads:

The QR Code is part of the tracking of our manufacturer. 
It is not intended to be used as product identification.
If you want to use it to identify your device:

```
0044 30.2019 023800 USBSDMux rev4
^^^^ - Add 299 to obtain the serial number
     ^^^^^^^ - This needs to match to make sure it is a USB-SD-Mux revision 4.
```

Does this information help you?

sephalon, 2019-11-18:
Thanks, that clarifies the situation. Closed.

closed 2019-11-18 by sephalon