linux-automation/usbsdmux #22: USB-SD-Mux within regular SD card slot

sephalon, 2020-01-29:
I am aware that the [manual](https://testbed.linux-automation.com/usbsdmux-M01/intended-usecase.html) clearly states that

> The USB-SD-Mux is intended to be directly inserted into the SD card slot of a DUT.

However, we still have quite a lot of boards that only have regular SD card slots, but no microSD one. I did some experimentation and realized that

- a microSD to SD card adapter,
- a SanDisk Ultra microSD card and,
- deactivating high speed mode (`broken-mmc-highspeed` device tree property)

seems to be a reliable combination to use the USB-SD-Mux within a regular SD card slot.

However, this is a far from ideal situation with respect to ESD and reliability. Hence, I would like to ask if you have any plans to release an alternative USB-SD-Mux hardware version for regular SD card slots? Or do you have any other ideas how this scenario could be supported?

SmithChart, 2020-01-31:
> I am aware that the [manual](https://testbed.linux-automation.com/usbsdmux-M01/intended-usecase.html) clearly states that
>
>
> However, we still have quite a lot of boards that only have regular SD [...]

Of course you are free to do what ever you want with your device. To be honest: we have a few devices in our lab where we need to use a adapter or an extension cable. The thing is: as a manufacturer we can not guarantee a good EMC-behavior in this cases. 

We currently have no plans to release such a USB-SD-Mux.

Thinking theoretically what would you want it to look like? Full-Size shape but with a Micro-SD socket or full-size shape with a full-size socket?

sephalon, 2020-01-31:
> To be honest: we have a few devices in our lab where we need to use a adapter or an extension cable.

That is good to know :-) Would you like to share which combination of SD cards and adapter/extension cables are working well for you (also by private email if you do not want to put advertisements for specific manufacturers/products here)? Have been able to use higher speed modes with such a configuration?

> Thinking theoretically what would you want it to look like? Full-Size shape but with a Micro-SD socket or full-size shape with a full-size socket?

Full size shape with microSD socket. I do not see any benefit in having a full size socket, it would only make the USB-SD-Mux less versatile by increasing its size.

> Or do you have any other ideas how this scenario could be supported?

I do not know enough about PCB design to comment on that, but if it works, we would be interested too :-)

SmithChart, 2020-02-02:
> That is good to know :-) Would you like to share which combination of SD cards and adapter/extension cables are working well for you (also by private email if you do not want to put advertisements [...]

I will have a look around our lab what setups we are currently using.
If you drop me an E-Mail I can send you some products we are currently using.

> Full size shape with microSD socket. I do not see any benefit in having a full size socket, it would only make the USB-SD-Mux less versatile by increasing its size.

That is what I would suggest, too. In my understanding both cards are the same internally.

sephalon, 2020-02-12:
> I will have a look around our lab what setups we are currently using.
> If you drop me an E-Mail I can send you some products we are currently using.

Thanks, dropped you an email.

I suggest we leave this issue open for further reference.

SmithChart, 2020-03-06:
> I will have a look around our lab what setups we are currently using.
> If you drop me an E-Mail I can send you some products we are currently using.

I did a short survey through our lab.
I found: 12 Devices with Micro-SD-Slots, but only 4 devices with full-size SD-Slots.
The combinations of adapters and cards seems totally random. Most cards seem to be from Intenso (Class 4 and Class 10) though.

PR #35 (2021-04-08): Document known issues of the product in the README
There are known issues regarding new high-speed devices like
the Raspberry Pi 4 that should be documented more publicly,
including possible workarounds.

Documenting these shortcomings here closes #25.

Another issue, that boil down to the same signal-integrity
problems posed by additional load on the data lines,
is the use of the USB-SD-Mux in full-sized SD-Card slots.

Documenting these shortcomings here closes #22.
commit 7deea419 (2021-04-08): Document known issues of the product in the README

There are known issues regarding new high-speed devices like
the Raspberry Pi 4 that should be documented more publicly,
including possible workarounds.

Documenting these shortcomings here closes #25.

Another issue, that boil down to the same signal-integrity
problems posed by additional load on the data lines,
is the use of the USB-SD-Mux in full-sized SD-Card slots.

Documenting these shortcomings here closes #22.
closed 2021-04-08 by hnez