linux-automation/usbsdmux #25: DUT (RP 4) doesn't detect the sdcard

atishp04, 2020-07-30:
Hi,
I bought two USB-SD-Mux recently. I install usbsdmux and switched to host mode. It worked fine.
After I switched it to DUT mode, the  test board (a Raspberry PI 4) doesn't detect the sdcard.
I can see the correct leds are on while switching between host and DUT mode.

I have tried several sdcards in both the USB-SD-Mux. The board detects the sdcard if inserted directly and works fine without any issues. Any suggestions what might be wrong ?

atishp04, 2020-07-31:
ping ?

SmithChart, 2020-07-31:
Hi atishp04,

we have seen some issues on some Embedded Devices where the ROM-Code is not able to detect an SD-Card in the USB-SD-Mux that otherwise works fine in a card reader.

Could you give it a try to switch the USB-SD-Mux into DUT mode and insert it into a normal cardreader? Just to make sure that we are not chaing an issue with the USB-SD-Mux itself.

I'll fetch one of our RPi 4 and give it a try myself.

Chris

SmithChart, 2020-07-31:
Resulsts from my bench test:

Raspberry Pi 4B, USB-SD-Mux
Micro SD Class 10 (Verbatim) works
Micro SD UHS 1 (San Disk Ultra) does not work.

So maybe using a slower Micro-SD Card can mitigate your problem. I will continue to investigate the problem with the UHS 1 card.

atishp04, 2020-07-31:
@SmithChart Thanks for the testing. It also doesn't work with the slowest one (SD class 4).

I have tried these sdcards.

SanDisk Micro SD HC I (sd class 4) [1]
SanDisk Micro SD XC1 (USH I) - SanDisk extreme

Unfortunately, I don't have Micro SD Class 10 with me to test it. I will try to buy one and test.
Is the Micro SD Class 10 only recommended class that works ? As per the sandisk guide, SanDisk High Endurance microSD is the only one with sd class 10 speed :(.

[1] https://kb.sandisk.com/app/answers/detail/a_id/22697/~/microsd%2Fmicrosdhc%2Fmicrosdxc-card-support-information-page

SmithChart, 2020-08-03:
The USB-SD-Mux should work with all speed classes. But the USB-SD-Mux introduces some additional capacitive and resistive load to the singals. Depending on the drive strength of the SD-Card and the SD-Card host this can lead to failing communications.

I plan to take a closer looks on the electrical effects here. I have a few RPi 4 laying around. So making measurements should be no problem. I'll let you know once I found anything.

SmithChart, 2020-08-10:
I have done some more tests:

First:
The USB-SD-Mux works with my Class 10 card if I swap R101 and R102 with 0 Ohms (instead of the original 33 Ohms).

Second:
I activated the bootloader's serial output using this document: https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md

For a failing boot I get:
```
PM_RSTS: 0x00001000
RPi: BOOTLOADER release VERSION:a5e1b95f DATE: Apr 16 2020 TIME: 18:11:29 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1587057086 0xad4240bb 0x00c03111
uSD voltage 3.3V
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
CID: 000353445343333247804d2b11ec012a
CSD: 400e00325b590000edc87f800a404000
SD: bus-width: 4 spec: 2 SCR: 0x02358043 0x00000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
SD CMD: 0x371a0010 (55) 0xaaaa0000 0x1fff0000
Failed to open device: 'sdcard'
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
CID: 000353445343333247804d2b11ec012a
CSD: 400e00325b590000edc87f800a404000
SD: bus-width: 4 spec: 2 SCR: 0x02358043 0x00000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
SD CMD: 0x371a0010 (55) 0xaaaa0000 0x1fff0000
Failed to open device: 'sdcard'
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
CID: 000353445343333247804d2b11ec012a
CSD: 400e00325b590000edc87f800a404000
SD: bus-width: 4 spec: 2 SCR: 0x02358043 0x00000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
SD CMD: 0x371a0010 (55) 0xaaaa0000 0x1fff0000
Failed to open device: 'sdcard'
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
CID: 000353445343333247804d2b11ec012a
CSD: 400e00325b590000edc87f800a404000
SD: bus-width: 4 spec: 2 SCR: 0x02358043 0x00000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
SD CMD: 0x371a0010 (55) 0xaaaa0000 0x1fff0000
Failed to open device: 'sdcard'
SD boot failed
BOOT_STOPPED: 0x00000001 0x0
FATAL @ 0x80001fde 0x00000004 0x800016e4
Initialising SDRAM 'Micron' 16Gb x2 total-size: 32 Gbit 3200
```

For a good boot I get:
```
PM_RSTS: 0x00001000
RPi: BOOTLOADER release VERSION:a5e1b95f DATE: Apr 16 2020 TIME: 18:11:29 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1587057086 0xad4240bb 0x00c03111
uSD voltage 3.3V
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
CID: 000353445343333247804d2b11ec012a
CSD: 400e00325b590000edc87f800a404000
SD: bus-width: 4 spec: 2 SCR: 0x02358043 0x00000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
SD CMD: 0x113a0010 (17) 0x0 0x1fff0202
Failed to open device: 'sdcard'
SD error: disable high speed
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD CMD: 0x371a0010 (55) 0xaaaa0000 0x1fff0001
Failed to open device: 'sdcard'
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
CID: 000353445343333247804d2b11ec012a
CSD: 400e00325b590000edc87f800a404000
SD: bus-width: 4 spec: 2 SCR: 0x02358043 0x00000000
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 12500000 Hz actual: 12500000 HZ div: 20 (10) status: 0x1fff0000 delay: 8
MBR: 0x00002000,  524288 type: 0x0c
MBR: 0x00082000,61801472 type: 0x83
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1 r-dir 2 r-sec 0
WEL: 0x00003fa2 0x00081fff
autoboot.txt not found
PM_RSTS: 0x00001000
Partition: 0
lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1 r-dir 2 r-sec 0
Read config.txt bytes     1798 hnd 0x0001976b hash fd77ea344d50ff36
Initialising SDRAM 'Micron' 16Gb x2 total-size: 32 Gbit 3200
recover4.elf not found
recovery.elf not found
Read start4.elf bytes  2252416 hnd 0x00010108 hash 604055cff3214d78
Read fixup4.dat bytes     5400 hnd 0x000002e9 hash a29f862f2cffc816
0x00c03111 0x00000000 0x0000000f
MEM GPU: 76 ARM: 948 TOTAL: 1024
FIXUP src: 128 256 dst: 948 1024
Starting start4.elf @ 0xfec00200

[    0.000000] Booting Linux on physical CPU 0x0
```

But I wasn't able to verify the difference in electrical signalling with my Oscilloscope.

atishp04, 2020-08-10:
> if I swap R101 and R102 with 0 Ohms

Thanks for testing. Does that mean sd class 10 doesn't work without replacing R101 and R102 ?
That would mean that usbsdmux doesn't work with any sdcard by default ?

atishp04, 2020-10-14:
Any update on this ?

PR #35 (2021-04-08): Document known issues of the product in the README
There are known issues regarding new high-speed devices like
the Raspberry Pi 4 that should be documented more publicly,
including possible workarounds.

Documenting these shortcomings here closes #25.

Another issue, that boil down to the same signal-integrity
problems posed by additional load on the data lines,
is the use of the USB-SD-Mux in full-sized SD-Card slots.

Documenting these shortcomings here closes #22.
commit 7deea419 (2021-04-08): Document known issues of the product in the README

There are known issues regarding new high-speed devices like
the Raspberry Pi 4 that should be documented more publicly,
including possible workarounds.

Documenting these shortcomings here closes #25.

Another issue, that boil down to the same signal-integrity
problems posed by additional load on the data lines,
is the use of the USB-SD-Mux in full-sized SD-Card slots.

Documenting these shortcomings here closes #22.
closed 2021-04-08 by hnez