linux-automation/usbsdmux #29: Installation on ubuntu-1804 does not install usbsdmux

DennisLipschinski, 2020-12-09:
If I run the installation/build as documented:
```
$ python3 -m venv venv
$ source venv/bin/activate
$ python setup.py install
$ sudo usbsdmux /dev/sg1 dut
```
I do not get the usbsdmux executable:
```
bash: usbsdmux: command not found
```
Can not find the file in venv/bin.

SmithChart, 2021-03-29:
Hey @DennisLipschinski,

sorry I can not reproduce your issue on my Ubuntu 18.04:

I installed the usb-sd-mux with the system python:

```
$ python3 -m venv testenv
$ source testenv/bin/activate
$ python setup.py install
```

As non root i can now use the usbsdmux command directly from the venv:
```
$ usbsdmux
usage: usbsdmux [-h] [-d] [-c] [-s SOCKET] SG {dut,host,off,client}
usbsdmux: error: the following arguments are required: SG, mode
```

As sudo I need to use the wrapper from the venv:
```
$ ls ./testenv/bin/usbsdmux -lah
-rwxrwxr-x 1 chris chris 268 Mär 29 09:13 ./testenv/bin/usbsdmux
$ sudo ./testenv/bin/usbsdmux
[sudo] password for chris: 
usage: usbsdmux [-h] [-d] [-c] [-s SOCKET] SG {dut,host,off,client}
usbsdmux: error: the following arguments are required: SG, mode
```

Are you sure that it's not present in your venv? The generic exclude from #30 should not be a problem here.

hnez, 2021-04-08:
Hi,

we have updated this projects README so that it contains a working example on how to use `usbsdmux` with `sudo`.
I consider this a solution to this issue, so i will close this issue for now.

Feel free to contact us again, should you have any further issues
Leonard

closed 2021-04-08 by hnez

DennisLipschinski, 2021-04-12:
Thanks for your feedback.

However I still have the problems:
```
sick@ubuntu:~/projects$ git clone https://github.com/linux-automation/usbsdmux.git usbsdmuxtest
Cloning into 'usbsdmuxtest'...
remote: Enumerating objects: 85, done.
remote: Counting objects: 100% (85/85), done.
remote: Compressing objects: 100% (54/54), done.
remote: Total 342 (delta 47), reused 60 (delta 27), pack-reused 257
Receiving objects: 100% (342/342), 96.83 KiB | 0 bytes/s, done.
Resolving deltas: 100% (193/193), done.
Checking connectivity... done.
sick@ubuntu:~/projects$ cd usbsdmuxtest/
sick@ubuntu:~/projects/usbsdmuxtest$ python3 -m venv venv
sick@ubuntu:~/projects/usbsdmuxtest$ source venv/bin/activate
(venv) sick@ubuntu:~/projects/usbsdmuxtest$ python setup.py install
fastentrypoints
running install
running bdist_egg
running egg_info
creating usbsdmux.egg-info
writing usbsdmux.egg-info/PKG-INFO
writing dependency_links to usbsdmux.egg-info/dependency_links.txt
writing top-level names to usbsdmux.egg-info/top_level.txt
writing entry points to usbsdmux.egg-info/entry_points.txt
writing manifest file 'usbsdmux.egg-info/SOURCES.txt'
reading manifest file 'usbsdmux.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no previously-included files matching '*~' found anywhere in distribution
warning: no previously-included files matching '*.swp' found anywhere in distribution
warning: no previously-included files matching '*.pyc' found anywhere in distribution
writing manifest file 'usbsdmux.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib
creating build/lib/usbsdmux
copying usbsdmux/pca9536.py -> build/lib/usbsdmux
copying usbsdmux/usb2642eeprom.py -> build/lib/usbsdmux
copying usbsdmux/ctypehelper.py -> build/lib/usbsdmux
copying usbsdmux/usbsdmux.py -> build/lib/usbsdmux
copying usbsdmux/__init__.py -> build/lib/usbsdmux
copying usbsdmux/__main__.py -> build/lib/usbsdmux
copying usbsdmux/service.py -> build/lib/usbsdmux
copying usbsdmux/usb2642i2c.py -> build/lib/usbsdmux
creating build/bdist.linux-x86_64
creating build/bdist.linux-x86_64/egg
creating build/bdist.linux-x86_64/egg/usbsdmux
copying build/lib/usbsdmux/pca9536.py -> build/bdist.linux-x86_64/egg/usbsdmux
copying build/lib/usbsdmux/usb2642eeprom.py -> build/bdist.linux-x86_64/egg/usbsdmux
copying build/lib/usbsdmux/ctypehelper.py -> build/bdist.linux-x86_64/egg/usbsdmux
copying build/lib/usbsdmux/usbsdmux.py -> build/bdist.linux-x86_64/egg/usbsdmux
copying build/lib/usbsdmux/__init__.py -> build/bdist.linux-x86_64/egg/usbsdmux
copying build/lib/usbsdmux/__main__.py -> build/bdist.linux-x86_64/egg/usbsdmux
copying build/lib/usbsdmux/service.py -> build/bdist.linux-x86_64/egg/usbsdmux
copying build/lib/usbsdmux/usb2642i2c.py -> build/bdist.linux-x86_64/egg/usbsdmux
byte-compiling build/bdist.linux-x86_64/egg/usbsdmux/pca9536.py to pca9536.cpython-35.pyc
byte-compiling build/bdist.linux-x86_64/egg/usbsdmux/usb2642eeprom.py to usb2642eeprom.cpython-35.pyc
byte-compiling build/bdist.linux-x86_64/egg/usbsdmux/ctypehelper.py to ctypehelper.cpython-35.pyc
byte-compiling build/bdist.linux-x86_64/egg/usbsdmux/usbsdmux.py to usbsdmux.cpython-35.pyc
byte-compiling build/bdist.linux-x86_64/egg/usbsdmux/__init__.py to __init__.cpython-35.pyc
byte-compiling build/bdist.linux-x86_64/egg/usbsdmux/__main__.py to __main__.cpython-35.pyc
byte-compiling build/bdist.linux-x86_64/egg/usbsdmux/service.py to service.cpython-35.pyc
byte-compiling build/bdist.linux-x86_64/egg/usbsdmux/usb2642i2c.py to usb2642i2c.cpython-35.pyc
creating build/bdist.linux-x86_64/egg/EGG-INFO
copying usbsdmux.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating dist
[... 68 lines cut ...]
creating build/bdist.linux-x86_64/egg/EGG-INFO
copying usbsdmux.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
copying usbsdmux.egg-info/SOURCES.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying usbsdmux.egg-info/dependency_links.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying usbsdmux.egg-info/entry_points.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying usbsdmux.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating 'dist/usbsdmux-0.2.0-py3.5.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
Processing usbsdmux-0.2.0-py3.5.egg
Removing /home/sick/projects/usbsdmuxtest/venv/lib/python3.5/site-packages/usbsdmux-0.2.0-py3.5.egg
Copying usbsdmux-0.2.0-py3.5.egg to /home/sick/projects/usbsdmuxtest/venv/lib/python3.5/site-packages
usbsdmux 0.2.0 is already the active version in easy-install.pth
Installing usbsdmux-service script to /home/sick/projects/usbsdmuxtest/venv/bin
Installing usbsdmux script to /home/sick/projects/usbsdmuxtest/venv/bin
Installing usbsdmux-configure script to /home/sick/projects/usbsdmuxtest/venv/bin

Installed /home/sick/projects/usbsdmuxtest/venv/lib/python3.5/site-packages/usbsdmux-0.2.0-py3.5.egg
Processing dependencies for usbsdmux==0.2.0
Finished processing dependencies for usbsdmux==0.2.0
(venv) sick@ubuntu:~/projects/usbsdmuxtest$ usbsdmux
usage: usbsdmux [-h] SG {get,dut,client,host,off}
usbsdmux: error: the following arguments are required: SG, mode
(venv) sick@ubuntu:~/projects/usbsdmuxtest$ 
```

hnez, 2021-04-12:
Hey!

I am sorry to hear that our software still is not working for you. To track down the problem i went and grabbed an Ubuntu 18.04 from the internet.

Here is a (somewhat abbreviated) list of commands that I executed in the Ubuntu live environment:

```
ubuntu@ubuntu:~$ uname -a
Linux ubuntu 5.4.0-42-generic #46~18.04.1-Ubuntu SMP Fri Jul 10 07:21:24 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
ubuntu@ubuntu:~$ lsb_release -a
No LSB modules are available.
Distributor ID:        Ubuntu
Description:        Ubuntu 18.04.5 LTS
Release:        18.04
Codename:        bionic

ubuntu@ubuntu:~$ sudo nano /etc/apt/sources.list # to add universe sources

ubuntu@ubuntu:~$ sudo apt update

ubuntu@ubuntu:~$ sudo apt install git pyhton3-venv

ubuntu@ubuntu:~$ Downloads/

ubuntu@ubuntu:~/Downloads$ git clone https://github.com/linux-automation/usbsdmux.git usbsdmuxtest
Cloning into 'usbsdmuxtest'...
remote: Enumerating objects: 85, done.
remote: Counting objects: 100% (85/85), done.
remote: Compressing objects: 100% (54/54), done.
remote: Total 342 (delta 47), reused 60 (delta 27), pack-reused 257
Receiving objects: 100% (342/342), 96.83 KiB | 4.21 MiB/s, done.
Resolving deltas: 100% (193/193), done.

ubuntu@ubuntu:~/Downloads$ cd usbsdmuxtest/

ubuntu@ubuntu:~/Downloads/usbsdmuxtest$ python3 -m venv venv

ubuntu@ubuntu:~/Downloads/usbsdmuxtest$ source venv/bin/activate

(venv) ubuntu@ubuntu:~/Downloads/usbsdmuxtest$ python
Python 3.6.9 (default, Jan 26 2021, 15:33:00) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import setuptools
>>> setuptools.__version__
'39.0.1'

(venv) ubuntu@ubuntu:~/Downloads/usbsdmuxtest$ python setup.py install
... Some lines removed ...
Installing usbsdmux script to /home/ubuntu/Downloads/usbsdmuxtest/venv/bin
Installing usbsdmux-configure script to /home/ubuntu/Downloads/usbsdmuxtest/venv/bin
Installing usbsdmux-service script to /home/ubuntu/Downloads/usbsdmuxtest/venv/bin

Installed /home/ubuntu/Downloads/usbsdmuxtest/venv/lib/python3.6/site-packages/usbsdmux-0.2.0-py3.6.egg
Processing dependencies for usbsdmux==0.2.0
Finished processing dependencies for usbsdmux==0.2.0

(venv) ubuntu@ubuntu:~/Downloads/usbsdmuxtest$ ls venv/bin/
activate  activate.csh  activate.fish  easy_install  easy_install-3.6  pip  pip3  pip3.6  python  python3  usbsdmux  usbsdmux-configure  usbsdmux-service

(venv) ubuntu@ubuntu:~/Downloads/usbsdmuxtest$ usbsdmux --help
usage: usbsdmux [-h] SG {get,dut,client,host,off}

positional arguments:
  SG                    /dev/sg* to use
  {get,dut,client,host,off}
                        Action:
                        get - return selected mode
                        dut - set to dut mode
                        client - set to dut mode (alias for dut)
                        host - set to host mode
                        off - set to off mode

optional arguments:
  -h, --help            show this help message and exit
```

As you can see I was not able to reproduce the issues you saw. Could you please check if the version I've seen in my run match the version numbers seen in you installation?

Greetings from Hildesheim
Leonard