linux-automation/usbsdmux #31: Device paths when using multiple usbsdmuxs

HerrMuellerluedenscheid, 2021-01-14:
Hi,
I'm currently using two (and soon three) usbsdmux in our CI pipeline. When switching to `host` mode, how can I know (despite manually checking the output of e.g. `dmesg`) where the device is mounted (`/dev/sda`, `/dev/sdb`, etc.)? Can I explicitly set a mount point?

Best regards
Marius

Bastian-Krause, 2021-01-14:
Hey,

the sg devices can be identified (with the [shipped udev rule](https://github.com/linux-automation/usbsdmux#reliable-names-for-the-usb-sd-mux)) via the serial number as `/dev/usb-sd-mux/id-*` as well as the storage device (mounting is the wrong term here) as `/dev/disk/by-id/usb-LinuxAut_sdmux_HS-SD_MMC_*-0:0`.

E.g.:
```
bst@test:~$ ls -la /dev/usb-sd-mux/id-000000000378 
lrwxrwxrwx 1 root root 6 Jan 10 02:23 /dev/usb-sd-mux/id-000000000378 -> ../sg1
bst@test:~$ ls -la /dev/disk/by-id/usb-LinuxAut_sdmux_HS-SD_MMC_000000000378-0\:0
lrwxrwxrwx 1 root root 9 Jan 10 02:23 /dev/disk/by-id/usb-LinuxAut_sdmux_HS-SD_MMC_000000000378-0:0 -> ../../sdb

```

HerrMuellerluedenscheid, 2021-01-14:
Awesome!

closed 2021-01-14 by HerrMuellerluedenscheid