linux-automation/usbsdmux #44: REST API

masselstine, 2022-11-02:
Hi folks,

Thanks for producing the sd-mux and this useful project to manage the device(s). I have written a REST API to allow a central machine host one or more devices which can then allow users or automation to use the mux without requiring accounts etc... with basic calls such as:

```
curl <host_name_or_ip>:5000/get/<mux_id>
curl <host_name_or_ip>:5000/set/<mux_id>/off
curl -X POST <host_name_or_ip>:5000/write/<mux_id> -F 'file=@test.img.gz'
curl <host_name_or_ip>:5000/task/<mux_id>
```
I haven't posted the code on github yet as I need to do a bit of cleanup first. I think the code would do better here though then on my personal github account. Would you be interested in the code? It is fully functional, though maybe not the prettiest, but definitely a good starting point.

SmithChart, 2022-11-07:
Hey @masselstine,

that's sounds like an interesting solution. In our infrastructure we rely on labgrid and that needs SSH anyway. But I can totally imagine that this can be useful. So you have a service running on the machine where the USB-SD-Mux is connected to and the machine controlling the devices make a HTTP-request to this machine?

I am not sure if our repo is the best place for your tool. We will probably not use your tool ourself but would still have to maintain it. This is probably not the best solution.
But we can definitely add a link to your tool to our readme and/or handbook. Would you mind to publish the code?

masselstine, 2022-11-07:
@SmithChart thanks so much for the reply.

Correct, you can make use of curl, http(ie), or similar, with no security at this point.

Completely understandable position, I appreciate you getting back to me with your thoughts.

What I might do in addition to getting the code cleaned up, is to get it posted as a company repository instead of under my account. This would probably be a better place to mention or link to. Please feel free to close this issue and when I have things ready (which will be a few weeks as I have some upcoming travel and such) I will send a pull request that you can do with as you please.

Thanks again and thanks for your time and contributions.

closed 2022-11-07 by masselstine

d21d3q, 2024-01-13:
I would recommend using https://nodered.org/  
You can use it to execute shell commands (run flash script, set DUT), control GPIO (of rpi, with relay which controls power for device being tested), and easily create [dashboard](https://flows.nodered.org/node/node-red-dashboard) with some buttons, status: 
[image]
(buttons on screen shot have labels "download dev image", "flash dev image", "download and flash")