linux-automation/usbsdmux #45: Arch Linux : OSError: [Errno 123] No medium found: '/dev/sda'

lucastanure, 2022-12-04:
`tanureal@cryzen usbsdmux $ source venv/bin/activate 

(venv) tanureal@cryzen usbsdmux $ sudo venv/bin/usbsdmux /dev/sda get
Traceback (most recent call last):
  File "/home/tanureal/workspace/usbsdmux/venv/bin/usbsdmux", line 10, in <module>
    sys.exit(main())
  File "/home/tanureal/workspace/usbsdmux/venv/lib/python3.10/site-packages/usbsdmux-0.2.1-py3.10.egg/usbsdmux/__main__.py", line 97, in main
  File "/home/tanureal/workspace/usbsdmux/venv/lib/python3.10/site-packages/usbsdmux-0.2.1-py3.10.egg/usbsdmux/__main__.py", line 79, in main
  File "/home/tanureal/workspace/usbsdmux/venv/lib/python3.10/site-packages/usbsdmux-0.2.1-py3.10.egg/usbsdmux/usbsdmux.py", line 55, in get_mode
  File "/home/tanureal/workspace/usbsdmux/venv/lib/python3.10/site-packages/usbsdmux-0.2.1-py3.10.egg/usbsdmux/pca9536.py", line 73, in read_register
  File "/home/tanureal/workspace/usbsdmux/venv/lib/python3.10/site-packages/usbsdmux-0.2.1-py3.10.egg/usbsdmux/usb2642i2c.py", line 413, in write_read_to
  File "/home/tanureal/workspace/usbsdmux/venv/lib/python3.10/site-packages/usbsdmux-0.2.1-py3.10.egg/usbsdmux/usb2642i2c.py", line 335, in _call_IOCTL
OSError: [Errno 123] No medium found: '/dev/sda'

(venv) tanureal@cryzen usbsdmux $ ls -l /dev/sda 
brw-rw---- 1 777 disk 8, 0 Dec  4 10:43 /dev/sda
`

lucastanure, 2022-12-04:
usbsdmux doesn't work on Arch Linux.
The device /dev/sda is created when I insert the USB cable into my host machine, but usbsdmux doesn't talk to it.

Bastian-Krause, 2022-12-04:
usbsdmux talks to the SCSI generic device `/dev/sgX`, not `/dev/sdX`. See the README and the help text.

lucastanure, 2022-12-09:
on arch linux kernel 6.0.11-arch1-1 the insertion of usbsdmux board doesn't create /dev/sgX devices.
Only /dev/sda.
[ 9238.242294] usb 7-3.3: new high-speed USB device number 8 using xhci_hcd
[ 9238.333687] usb 7-3.3: New USB device found, idVendor=0424, idProduct=2640, bcdDevice= 8.a2
[ 9238.333694] usb 7-3.3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 9238.388888] hub 7-3.3:1.0: USB hub found
[ 9238.389078] hub 7-3.3:1.0: 1 port detected
[ 9238.668937] usb 7-3.3.1: new high-speed USB device number 9 using xhci_hcd
[ 9238.818296] usb 7-3.3.1: New USB device found, idVendor=0424, idProduct=4041, bcdDevice= 2.09
[ 9238.818304] usb 7-3.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9238.818307] usb 7-3.3.1: Product: usb-sd-mux_rev4.0
[ 9238.818309] usb 7-3.3.1: Manufacturer: Linux Automation GmbH
[ 9238.818310] usb 7-3.3.1: SerialNumber: 000000001323
[ 9238.818771] usb-storage 7-3.3.1:1.0: USB Mass Storage device detected
[ 9238.819079] scsi host2: usb-storage 7-3.3.1:1.0
[ 9239.839308] scsi 2:0:0:0: Direct-Access     LinuxAut sdmux HS-SD/MMC  2.09 PQ: 0 ANSI: 0
[ 9239.840213] sd 2:0:0:0: [sda] Media removed, stopped polling
[ 9239.841102] sd 2:0:0:0: [sda] Attached SCSI removable disk

hnez, 2022-12-09:
Hi, that's strange.
I've just plugged an USB-SD-Mux into my Arch Machine and for me everything looks fine:

    leonard@fulladder ~> uname -a
    Linux fulladder 6.0.10-arch2-1 #1 SMP PREEMPT_DYNAMIC Sat, 26 Nov 2022 16:51:18 +0000 x86_64 GNU/Linux

    leonard@fulladder ~> sudo dmesg | tail -n 15
    [151042.487522] usb 1-5: New USB device strings: Mfr=0, Product=0, SerialNumber=0
    [151042.492601] hub 1-5:1.0: USB hub found
    [151042.495491] hub 1-5:1.0: 1 port detected
    [151042.795375] usb 1-5.1: new high-speed USB device number 6 using xhci_hcd
    [151042.963585] usb 1-5.1: New USB device found, idVendor=0424, idProduct=4041, bcdDevice= 2.09
    [151042.963594] usb 1-5.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
    [151042.963598] usb 1-5.1: Product: usb-sd-mux_rev4.0
    [151042.963601] usb 1-5.1: Manufacturer: Linux Automation GmbH
    [151042.963604] usb 1-5.1: SerialNumber: 000000000757
    [151042.971619] usb-storage 1-5.1:1.0: USB Mass Storage device detected
    [151042.971873] scsi host10: usb-storage 1-5.1:1.0
    [151043.981281] scsi 10:0:0:0: Direct-Access     LinuxAut sdmux HS-SD/MMC  2.09 PQ: 0 ANSI: 0
    [151043.981586] sd 10:0:0:0: Attached scsi generic sg3 type 0
    [151043.982102] sd 10:0:0:0: [sdd] Media removed, stopped polling
    [151043.982667] sd 10:0:0:0: [sdd] Attached SCSI removable disk

    leonard@fulladder ~> ls /dev/sg*
    /dev/sg0  /dev/sg1  /dev/sg2  /dev/sg3

I am not currently running the newest Kernel, as I did not come around to rebooting yet.
This leads to problems when I have already `pacman -Syuu`-ed but not yet rebooted, as this results in the kernel modules for the currently running kernel no longer being available in `/lib/modules`.
This should not be the case for you, as you are running the newest kernel, but could you check anyways if the `sg` module is available and loaded on your system?

    leonard@fulladder ~> lsmod | grep sg
    sg                     49152  0

    leonard@fulladder ~> sudo modprobe sd

lucastanure, 2022-12-09:
Hi,

I do have the module sg, but I had to manually insert it and now works

hnez, 2022-12-09:
Huh, okay. I'would have expected the `sg` module to get loaded automatically, but it seems this is not the case for you.
Do you see any other `/dev/sg` devices now that the `sg` module is loaded?
In my case I have two SATA disks and a SATA optical drive. Maybe one of these has triggered the loading of the `sg` module.
Maybe your system does not have any such legacy devices.

We do have this section in the Troubleshooting section of the README:

* In order for the `/dev/sg*` device to appear the `sg` kernel module needs to be loaded into the kernel. This is usually done 
  automatically by udev once the USB-SD-Mux is connected. To manually load the kernel module run `sudo modprobe sg`.

But I would prefer this manual step not to be required. (And for most users it is'nt)

lucastanure, 2022-12-09:
yes, I only have nvme drives. I think your cdrom is triggering the load of sg.

Do you see any other /dev/sg devices now that the sg module is loaded?
No

pslldq, 2024-12-09:
An alternative to manually running modprobe would be `echo sg > /etc/modules-load.d/usbsdmux.conf`
This loads the `sg` module on every boot. See https://wiki.archlinux.org/title/Kernel_module#systemd

SmithChart, 2025-06-24:
I'll close this issue now as it does not seem to originate in our tool. Feel free to re-open if any new questions arise.

closed 2025-06-24 by SmithChart