linux-automation/usbsdmux #46: Support GPIOs for DUT/Host Switching and Card Detect Signal Toggling

mjk-gh, 2023-04-01:
We are going to buy one or more USB-SD-Muxes anyway, but it would be the cherry on top, if you could support the following two things:

1) Let the user switch the microSD card between host and DUT using one (host/DUT) or two (host/DUT/off) GPIOs[1] as input. Even if it seems like just a minor convenience feature, it can speed up development significantly, depending on your development workflow and tests.
 
2) Control DUT's card-detect signal via a GPIO[1] as output. The usual way (and also present in our DUTs) seems to be that the DUT's CD signal is pulled high and, upon insertion of a microSD card, shorted to GND, so setting the GPIO to low upon switching the card to the DUT side should accommodate this feature easily.

If possible (via firmware update), implement both 1) and 2) via firmware, as this allows for quasi-autonomous (i.e. PC-free) operation of the USB-SD-Mux as a simple and quickly operable microSD card on/off switch.

[1] (preferrably exposed on a VIA or a trace, but we would also be willing to bend ultra-fine-pitch pins under the microscope)

SmithChart, 2023-04-03:
As I understand your use-cases 1) and 2) are two different scenarios (as for 2 you will need a USB connection and in 1 you want to avoid it).

The USB-SD-Mux does not have any firmware (at least none that we program): It consists of an USB card reader with a spare I2C-bus. We only program our USB identification into there.

If you do not mind to make modifications to your devices you can probably get the same result by soldering some connections to the USB-SD-Mux:

**Control via GPIO:**
Internally we have three GPIOs that power the switching:
* _Select:_ 0: Card reader is connected to the sd-card, 1: DUT is connected to the SD-Card
* _DAT_ENABLE:_ 0: Signals are connected to the SD-card, 1: Signals are disconnected
* _VCC_ENABLE:_ 0: Power is connected to the SD-card, 1: Power is disconnected

If you only power the USB-Mux up (but do not initialize it with the _usbsdmuxctl_-tool) the I2C port expander will be high impedance (with weak pull-ups).
You cold connect your signal to these pins and override the power expander from the outside:
* Select: U5, Pin 3
* DAT_ENABLE: U5, Pin 1
* VCC_ENABLE: U5, Pin 2

You are free to switch DAT_ENABLE and VCC_ENABLE in one step. Signal levels are 0V / 3.3V.
You can power the USB-SD-Mux via 5V on U1, Pin 3 or 3.3V on U1, Pin 2+4. GND is U1 Pin 1.

In our control software we always switch to the _off_-state first. Then set the _select_-signal and afterwards set _DAT_ENABLE_ and _VCC_ENABLE_: https://github.com/linux-automation/usbsdmux/blob/master/usbsdmux/usbsdmux.py#L85

**Control DUT Card Detect:**
In the USB-SD-Mux design de decided against a pin for the DUT's card detect. In most use-cases power-cycling the DUT is the easiest way to use a new image without any hardware modifications. And that's why we don't have a spare GPIO for this use.

The signals from the port expander (and our internal control signals) all have the wrong polarity for an active-low card detect signal.
But: The _DUT-Mode_ LED is driven via a low-side N-MOSFET. The FET is active every time the data- and power-switches connect the SD-card to the DUT. So this signal could work for a card detect.
The best place to connect to this signal is highlighted here:
[image]

(The MOSFET is an ON-semi FDC6401 (https://www.onsemi.com/pdf/datasheet/fdc6401n-d.pdf ) that should have plenty of headroom for an external circuit. )

Does this answer your questions?

mjk-gh, 2023-11-25:
@SmithChart: First of all, I apologize for the delayed response!

Secondly: Yes, your very detailled reply did answer my questions. Both hacks seem to be good enough for our purposes -- the project has been pushed on the backburner a while ago, so I could not yet test anything, but I will report back as soon as the project gets in focus again (which it should within the next few weeks, hopefully).

SmithChart, 2024-04-16:
I will close this for now. @mjk-gh feel free to re-open or comment if you have got any feedback.

closed 2024-04-16 by SmithChart