linux-automation/usbsdmux #51: Adding USB2640 support?

perezmeyer, 2023-09-04:
The [SDWire v1.4](https://wiki.tizen.org/SDWire) uses the Microchop USB2640, which is currently supported by another software. How hard would it be to support this version too? I'm not a pythonist, that's why I am asking.

SmithChart, 2023-09-05:
Hey @perezmeyer ,
as far as I understand the SDWire does use an FTDI (as an extra USB device) to control the switch. So supporting the SDWire would mean to interact with this FTDI.
On our design we use the auxiliary I2C on the USB2642 to control the device - so we have a somewhat different architecture.

So changes would be needed in the way we identify the device (we currently use the `/dev/sg*` -device associated with the SCSI-device. For the SDWire we would probably want to move to a serial-number based identification. This would probably mean to interact with `udev` to find the correct devices.
Next step would be to control the GPIO on the FDTI using some library.

We currently do not have plans to add support for the SDWire. But if this is something you are interested in: Feel free to contribute.

perezmeyer, 2023-09-05:
Hi @SmithChart !

Thanks a lot for the feedback. Clearly two different paths, so I think it is better to keep using the current tool for the moment. Again, thanks!!

closed 2023-09-05 by perezmeyer