linux-automation/usbsdmux #72: [Hardware Feature Request] make SD connector flexible (Flex-PCB)

2mkx, 2024-04-15:
Hi,
my software colleague just came over to me "he needs this to debug - will it fit?" - but the SD socket is not at the edge of the board I'm designing. Moving it to the edge will create a mess :-/
My proposal / request would be if you could use a Semi-Flexible PCB design and just leave a few millimeters of the SD connector tongue flexible - just to allow a slight bend.
The benefit for all other users would be the reduced stress onto the SD connector when a (compared to a simple SD card) PCB and heavy USB cable is added to the socket.
Thanks :)

SmithChart, 2024-04-16:
Hi @2mkx , thanks for the input!
We have thought about this. But until now always dismissed the idea as difficult to handle in practice. You would have to insert the *wobbly* front portion of the device into the socket. Not sure how long a stiffened flex-pcb will withstand the forces. Maybe we can do a prototype one day.

Another option would be to use extension cables. We have used these two: [Option 1](https://www.reichelt.de/de/en/microsd-to-microsd-extension-16-cm-delock-61870-p145835.html) [Option 2](https://www.reichelt.de/de/en/raspberry-pi-microsd-extension-flexible-50-cm-rpi-msd-fl50-p223612.html) (We are not affiliated with the shop. Those are just examples.). Both work in *some* situations. But especially fast SD-Card modes may be problematic.

Cheers,
Chris

2mkx, 2024-04-16:
Thank you for the fast answer :)
One idea with the "wobbly front" would be to add a rigid stub to it - the flex-pcb manufacturers can remove the adhesive partially between the rigid and flex regions. After milling there would remain a "handle" at the stiffend SD connector.
[image]
Thank you for the links, indeed, they are quite long and I feel the same regarding highspeed signals.

cheers