linux-automation/usbsdmux #81: Current drive??

akomakul, 2024-06-13:
Hi,
Could you please let me know how much current would drive the below pin in the image?

[image]

akomakul, 2024-06-13:
Thanks for the reply.

Our device has 2 modes, flashing mode to update software via SD card and normal mode boots via eMMC.  We are connecting SD card to PC to write new image and then connect to DUT. After connecting to DUT we want to switch to flashing mode. For doing this we want to control the pin on the device using an extra GPIO pin on USB-SD-Mux but as such option is not available and as you have explained in https://github.com/linux-automation/usbsdmux/issues/49 we want to use the above mentioned pin.

SmithChart, 2024-06-13:
OK, that sounds reasonable. Driving some logic level pin should be OK for this net. (As long as the open drain output + Pull-Up to 3.3V works for your DUT.)

SmithChart, 2024-07-23:
Did not hear anything back from you. Closing this as solved for now. Feel free to open again.

closed 2024-07-23 by SmithChart