linux-automation/usbsdmux #82: Using multiple SDMuxes

Bozka222, 2024-09-13:
Hello, 
I have a problem with using two SDMux devices together. I am using server with Debian GNU/Linux 12 (bookworm). There is no problem with using one SDMux device but while I connect the second one to the server, first one disappears. This works back and forth so I can always see only one SDMux device. I can see both devices with command $ usb_devices but not with command $ ls -l /dev/usb-sd-mux nor with $ ls -l /dev/disk/by-id. While trying same configuration on ubuntu everything works as it should and I can see both devices connected.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

SmithChart, 2024-09-13:
In principal we expect multiple USB-SD-Mux on a single test server to work. This is taken from one of our test-servers:
```
cfi@rlabA-srv:~$ ls -lah /dev/usb-sd-mux/*
lrwxrwxrwx 1 root root 6 Sep  3 17:21 /dev/usb-sd-mux/id-000000000047 -> ../sg1
lrwxrwxrwx 1 root root 6 Sep 12 08:29 /dev/usb-sd-mux/id-000000000107 -> ../sg8
lrwxrwxrwx 1 root root 6 Sep  3 17:21 /dev/usb-sd-mux/id-000000000116 -> ../sg5
lrwxrwxrwx 1 root root 6 Sep  3 17:21 /dev/usb-sd-mux/id-000000000378 -> ../sg3
lrwxrwxrwx 1 root root 6 Sep 12 08:29 /dev/usb-sd-mux/id-000000000380 -> ../sg6
lrwxrwxrwx 1 root root 6 Sep  3 17:21 /dev/usb-sd-mux/id-000000000595 -> ../sg2
lrwxrwxrwx 1 root root 6 Sep  3 17:21 /dev/usb-sd-mux/id-000000001316 -> ../sg4
lrwxrwxrwx 1 root root 6 Sep 12 08:28 /dev/usb-sd-mux/id-000000001566 -> ../sg7
```

So this is definitely worth investigating.

What do you see in `dmesg -w` while plugging the first and then the second UBS-SD-Mux?

Bozka222, 2024-09-16:
After plugging first SD-Mux:
[image]
and second SD-Mux:
[image]

SmithChart, 2024-09-16:
Thanks. Each for itself looks OK: The only odd entry is the last line for the first sd-mux. 

Both devices use `/dev7sdb` and `/dev/sg1`, so assume there is something happening in between where these device nodes disappear?

Bozka222, 2024-09-16:
Yes they both use `/dev/sg1` and `/dev/sdb`. Another strange thing I noticed now is that they need to be connected both together. When I connect only one, I do not see any SDMux devices connected. When i connect both I can see only the one device which I connected last.

SmithChart, 2024-09-16:
What is happening inbetween? Can you provide the log? A `reset` should not be enough for the device to disappear. I would at least also suspect to see a `disconnect`.

Bozka222, 2024-09-16:
Well, there is nothing between. 
[image]

Bozka222, 2024-09-16:
OK I found the root cause. I have connected one of the USB devices to my virtual machine in proxmox so that's why one of them was disappearing. Anyway thanks for the help.

SmithChart, 2024-09-16:
Uh, that's an interesting effect. Good to hear if it works now.

I'll close this issue now. Feel free to reopen or open a new issue, if needed.

Cheers!

closed 2024-09-16 by SmithChart