linux-automation/usbsdmux #83: GPIO Pin's in usbsdmux are inverted

danieleftodi, 2024-10-30:
Hi, 

usbsdmux /dev/sg1 gpio 0 0
usbsdmux /dev/sg1 gpio 0 1

actually is Pin 1 on the physical board

and 

usbsdmux /dev/sg1 gpio 1 0
usbsdmux /dev/sg1 gpio 1 1

is Pin 0 on the physical board

Maybe invert the settings in usbsdmux, to reflect the Real world?

SmithChart, 2024-10-30:
[image]

SmithChart, 2024-10-30:
Oh, that's a nice one.
First: I can reproduce the issue.

The problems seems to originate in the schematic:
[image]
Next to `Extern GPIO` the GPIO signals are crossed for no apparent reason.
I don't understand why we did not catch during the schematic review - or at lest testing.

At least it is an easy fix in software. We already have a mapping in place:
https://github.com/linux-automation/usbsdmux/blob/35624f2848aaae8b2e60aa3f36cbe4629d4e8d00/usbsdmux/usbsdmux.py#L295

I will prepare an PR. Thanks for reporting!

SmithChart, 2024-11-05:
This issue should be fixed in [24.11](https://github.com/linux-automation/usbsdmux/releases/tag/24.11). Please give `24.11` or `24.11.1` a try.
Feel free to re-open if the issue persists.

And I think we have traced how this issue came to be:
It began with the swapped signal in the schematic (see above). But: Our initial prototypes did not have the fancy pin markings in silkscreen. So during software development we simply tested both outputs until one worked.
The silkscreen was added based on the net-names on the PCB.  So yeah. No one bothered to check this one more time :-)

closed 2024-11-05 by SmithChart