linux-automation/usbsdmux #84: [Feature Request] port usb-sd-mux to macOS

danieleftodi, 2024-10-30:
Hi, 

I recently bought a USB-SD-Mux FAST ... and I love it 😄 

... though, I noticed it doesn't have macOS support.

I have done some reading through your code and and the MicroChip USB2642 documentation. 

Especially the "[USB82642/2642 I2C Over USB User’s Guide - Using USB Mass Storage Class Bulk-Only Transport & SCSI Pass Through](https://ww1.microchip.com/downloads/en/DeviceDoc/50002283A.pdf)" 

From what I can gather, correct me if I'm wrong, but it just a matter of being able to send an I2C over USB - to the MicroChip USB2642.

Ergo, all we'd need to create is USB driver with DriverKit SDK for the usb-sd-mux.

And I do understand, that maybe you guy's don't have a Mac or then interested to do this work. But since this an OSS project. Could you maybe assist with some technical information, when needed? 

Cheers,
Daniel

[image]
[image]
[image]
[image]

Also found these sources:
[Building a Simple USB Driver](https://developer.apple.com/documentation/kernel/hardware_families/usb/building_a_simple_usb_driver)
[USBDriverKit](https://developer.apple.com/documentation/usbdriverkit)
[Creating a Driver Using the DriverKit SDK](https://developer.apple.com/documentation/driverkit/creating_a_driver_using_the_driverkit_sdk)
[Sending SCSI or ATA commands to storage devices](https://developer.apple.com/library/archive/qa/qa1179/_index.html)
[Accessing SCSI Architecture Model Devices](https://developer.apple.com/library/archive/documentation/DeviceDrivers/Conceptual/WorkingWithSAM/WWS_SAMDevInt/WWS_SAM_DevInt.html)
[WWDC 2020 - Modernize PCI and SCSI drivers with DriverKit](https://developer.apple.com/videos/play/wwdc2020/10210/)
[SCSIControllerDriverKit](https://developer.apple.com/documentation/scsicontrollerdriverkit)
[SCSIPeripheralsDriverKit](https://developer.apple.com/documentation/scsiperipheralsdriverkit)

hnez, 2024-10-30:
Hi,

> I recently bought a USB-SD-Mux FAST ... and I love it 😄

it's great to hear that, thanks!

> ... though, I noticed it doesn't have macOS support.

Also no Windows or BSD support. There are two reasons for that:

- We at Linux Automation GmbH and Pengutronix (our parent company) tend to be Linux fans.
- The [SCSI Generic driver](https://docs.kernel.org/scsi/scsi-generic.html) we use to communicate with the USB-SD-Mux is very Linux specific. Windows does seem to have a similar(-ish. At least as similar as Windows and Linux APIs get) API and various people have said they were interested in adding support for it to the USB-SD-Mux tool and we have also done some investigation ourself but we did not see anything get out of that yet.

While our own focus is very Linux-centric we would be very happy about contributions that add support for other platforms.

> I have done some reading through your code and and the MicroChip USB2642 documentation.
> 
> Especially the "[USB82642/2642 I2C Over USB User’s Guide - Using USB Mass Storage Class Bulk-Only Transport & SCSI Pass Through](https://ww1.microchip.com/downloads/en/DeviceDoc/50002283A.pdf)"
> 
> From what I can gather, correct me if I'm wrong, but it just a matter of being able to send an I2C message over USB - to the MicroChip USB2642.

I2C messages over SCSI over USB but yes, there is not much more to how the USB-SD-Mux works.

On the USB-SD-Mux FAST we have a TCA6408 i²c GPIO expander that is connected to the USB2642 via i²c and that controls the muxing and the external GPIOs.

To set the mux to HOST we set output pin 0 of the GPIO expander high.
To set the mux to DUT we set output pin 0 of the GPIO expander low.
To set the pin high/low we set/clear the corresponding bit in the GPIO expander output register via i²c:

https://github.com/linux-automation/usbsdmux/blob/35624f2848aaae8b2e60aa3f36cbe4629d4e8d00/usbsdmux/i2c_gpio.py#L91-L109

The i²c transfer is encapsulated in a SCSI command …

https://github.com/linux-automation/usbsdmux/blob/35624f2848aaae8b2e60aa3f36cbe4629d4e8d00/usbsdmux/usb2642.py#L425-L457

… the SCSI command is sent to the Linux Kernel via the SCSI generic device using an ioctl …

https://github.com/linux-automation/usbsdmux/blob/35624f2848aaae8b2e60aa3f36cbe4629d4e8d00/usbsdmux/usb2642.py#L325-L346

and the kernel wraps it into USB transfers.

> Ergo, all we would need to create, is a USB driver with DriverKit SDK for the usb-sd-mux.

I would suggest trying to implement it at the SCSI first, instead of the USB level.
Talking raw USB means having to implement the SCSI layer on top and more potential for confusing the USB and SCSI state machines in the USB2642 and your OS Kernel.

> And I do understand, that maybe you guy's don't have a Mac or interested to do this work. But since this an OSS project. Could you maybe assist with some technical information, when needed?

Sure! You can reach us here or in the #lxa channel on [libera.chat](https://libera.chat/).
We are usually not too secretive about the inner workings of our devices.

goldelico, 2026-09-24:
I investigated this feature request because I needed exactly this for a small mobile embedded-development setup.

We do embedded development and increasingly use cross-compilers on macOS to do Embedded Linux development (like our Embedded Linux system LetuxOS). Therefore, I wanted to put together a portable setup consisting of a MacBook M5 Pro, an USB hub, the USB MUX FAST and an FTDI interface, and the device under development. This meant that I needed to control the usbsdmux directly from macOS.

Unfortunately, the existing usbmux tool does not run on macOS as this request shows.

With some debugging help and conceptual analysis from ChatGPT, I eventually developed the attached patch after a few wrong turns. The approach is deliberately quite simple: detect the host OS and provide the corresponding implementation while leaving the existing Linux code essentially untouched.

It works for me. For example: sudo usbmux /dev/null host

Some implementation details: on macOS the MUX is detected directly via its USB bus ID rather than through SCSI (e.g. /dev/sg0) as on Linux. The main technical difference is that macOS does not see the device as a disk until it has been switched to host. Therefore, /dev/disk* or the filesystem cannot be used to communicate with the MUX. The macOS implementation has to communicate with it directly using USB packets using PyUSB, essentially reproducing the relevant Linux ioctl()functionality.

After having this working, however, I discovered another issue with my original overall setup: macOS has no built-in ext filesystem support, so it only mounts the FAT partitions of the µSD card. I also did not find an open-source solution that provides the required ext write support (macFUSE and ext4fuse alone is not sufficient). I did not want to buy and install the Paragon driver just for this purpose.

So I eventually changed the setup instead: I now run Debian 12 in a UTM VM on the Mac, with local root SSH access. The Linux usbsdmux runs there, so from macOS I can simply use ssh to switch the MUX and rsync to copy files to and from the µSD card.

Consequently, I don't actually need the macOS support for my own setup anymore. Nevertheless, I am posting the pull request (https://github.com/linux-automation/usbsdmux/pull/108) since the implementation works and may be useful for others who want to use usbsdmux directly from macOS.

Hopefully, this is useful for someone.