linux-automation/usbsdmux #87: Suggest to use udev for loading sg

frosteyes, 2025-01-28:
In the documentation it is suggested that sg is loaded using /etc/modules if it is not loaded when a mux is connected.
https://linux-automation.com/usbsdmux-M01/product-troubleshooting.html

Instead of always loading it - just use udev with a rule like below.
A udev rules is already created for plugdev - so just add it.

`ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="0424", ATTRS{idProduct}=="4041", RUN{builtin}+="kmod load sg"`

hnez, 2025-01-29:
Hi @frosteyes,

thanks for the suggestion! I've opened #88, which adds this change to [`contrib/udev/99-usbsdmux.rules`](https://github.com/linux-automation/usbsdmux/pull/88/files#diff-481ff2276696e1be436ff745e286492f2fd3361870df7128c0795d1c036e8e55), which we already recommend to install in the README.

The USB vendor and product IDs are not unique to the USB-SD-Mux, since they are the defaults for the SD Card reader IC we use on the USB-SD-Mux, so I've decided to match on `ATTRS{manufacturer}=="Linux Automation GmbH", ATTRS{product}=="usb-sd-mux*"` instead.

frosteyes, 2025-01-29:
Hi @hnez 

Thanks for quick reply, and cool with your patch. And good point with the matching ATTRS.
Will test it on Gentoo.

SmithChart, 2025-01-30:
I'll repoen this issue until I've updated the manual for the USB-SD-Mux.

SmithChart, 2025-04-22:
Added to the manual in [R13](https://linux-automation.com/usbsdmux-M01/index.html#term-usb-sd-mux-M01-R13-V01-2025-04-22).

PR #88 (2025-01-30): contrib/udev: explicitly load the sg module in the udev rule
We never got around to figuring out why the `sg` module is auto-loaded on some systems/distributions (e.g. Debian) but not on others (e.g. Arch). We have since used workarounds like adding `sg` to `/etc/modules`.

Instead just use udev to load the module if a new USB-SD-Mux is connected, as suggested by @frosteyes in https://github.com/linux-automation/usbsdmux/issues/87.

I've tested the change on may Arch Linux System that would not automatically load the `sg` module before, but does with the updated udev rule. I've also checked if [... cut]
closed 2025-04-22 by SmithChart