linux-automation/usbsdmux #98: Card detection very unreliable due to physical dimensions

marcowidmer, 2025-06-30:
In our projects, we use the DM3AT microSD Card socket. It has a simple physical switch to detect the card:

[image]

Unfortunately, on the USBSDMux (both "classic" and "fast" variants), the PCB corner which triggers the card detection switch is cut off:

[image]

This results in very unreliable card detection, as it depends on the exact insertion angle. If not inserted perfectly, U-Boot reports `MMC: no card present` and fails to boot the kernel.

SmithChart, 2025-07-24:
Hey @marcowidmer ,
thanks for reaching out!

The chamfer on the USB-SD-Mux outline (Both Classic and FAST) were added in an early prototype to ease the insertion into some Micro-SD sockets.
Thanks for the input that this feature is causing these issues for you. We will consider reduce or remove the chamfer in a future version.

For now I can only image two workarounds:

* We have [published the STEP-File](https://linux-automation.com/de/products/usb-sd-mux-fast.html) of the device. I know of customers that build custom jigs to hold the USB-SD-Mux firmly in place.
* At least on Linux you can disable the card detect in the devicetree using the `broken-cd` attribute.

(Interestingly: We are using the same DM3AT-Socket on our USB-SD-Muxes. But on the SD-Mux the card detect is not connected to anything. So we never noticed this issue.)

loic-AMCS, 2026-08-25:
Same problem here on a USB-SD-Mux FAST: the cut-off corner makes card detection unreliable, it depends on the exact angle the card sits at.

SmithChart, 2026-08-25:
We are currently preparing a new production run. So we want to take the chance and improve the outline.

I've taken an old USB-SD-Mux Classic prototype. That has the exact Micro-SD socket @marcowidmer has referenced in the first post and it is also the socket still used on the FAST. (Please ignore the adhesive residue on the socket and also the bad soldering where visible. This prototype has seen things over time... But the socket is still intact.)

# Reference: Standard Micro-SD-Card
This is how it looks with a Transcend 16GB Class 10 card: (This card is not in any way special. I've just had it lying on my desk.)
[image]
The card detect switch engages with the card by the fraction of a millimeter. And I guess that makes total sense: We want the contacts on the back to have solid contact to the card before we engage the card detect. So there simply isn't much headroom here.

# USB-SD-Mux FAST and Classic
This is how it looks like with an USB-SD-Mux Classic. The outline on the FAST is the same. So is true for both generations:
[image]
The card detect switch barely makes contact to the side of the USB-SD-Mux. It closes, as reported, when the USB-SD-Mux is inserted perfectly straight and fully into the DUT.
But once it slips a tiny bit out of the socket the switch actually helps to push the USB-SD-Mux out of the socket (or so it seems).

# What to do about it
So, what could be done about it? I would suggest to simply get rid of or at least reduce the chamfer on the front of the USB-SD-Mux.
As an example I've taken one of the earliest prototypes of the device. Here the front looks like this:
[image]
This has multiple problems in practice: The mouse-bites (both sides) and the latch (on top) do not work as expected.
But: It does not have the chamfer.

This is how it looks in the socket:
[image]
It engages well with the switch. Maybe even (a *wee bit*) better than the original Micro-SD card. 

So I guess this will be our way forward: We will reduce the chamfers on the front and simply live with the fact that the USB-SD-Mux may be a little harder to insert on the first try. But that is probably acceptable if it means that the device is more reliable in the long run.

I will update the PCB and do some tests with 3D-printed dummies and get back to you once I have some results to share.